maelstrom2001/ArcticFox #327: Disable AF logo on PowerOn and PowerOff animation

ReMicroN, 2017-05-08:
Hello! Can you add option in menu to disable AF logo on PowerOn and PowerOff animation. Thank You!

TBXin, 2017-05-08:
Hello,

You do not like animation or something like that?
I am just want to understand for what reason you want to disable they.

ReMicroN, 2017-05-08:
No. Animation is good. =) But I'm used to working screen immediately after "power on clicks" as on factory firmwares. So it will be good to make this option.

dannysmobiles, 2017-05-08:
can i say this is good as it is if want it like the factory firmware then use factory firmware

TBXin, 2017-05-08:
> So it will be good to make this option
You can just press Fire button once to skip welcome logo.

Extermicide, 2017-05-09:
No disrespect "RemicroN" (as I fully understand where your coming from with the request), but let's look at it from/with the NFE Team's eyes and viewpoint.

They worked very hard on this software, and give it to us for free.  This software is "ART", and like any artist, they would like to "sign" their hard work.   Being that's the only thing they seem to ask, it's not unreasonable to allow them to "sign" their work.

Cheer's NFE TEAM - and Thank You!

closed 2017-05-15 by TBXin