maelstrom2001/ArcticFox #328: watts

dannysmobiles, 2017-05-08:
hi just wonding when i set my watts to 70w but when i fire it goes down to 69.99 watts im just wonding why is that thank you and great work

TBXin, 2017-05-08:
Hello,

Q: The power goes down 0.1W when I press Fire, is that normal?
A: Yes, while pressing the Fire button, the mode shows the output power in real time.

maelstrom2001, 2017-05-08:
It is normal behavior. Resistance is measured in real time, it may fluctuate a little, which produces 0.1..0.2 W offset, 'cause current changed a little for this reason.

dannysmobiles, 2017-05-08:
thank you both of your i was not saying owt bad about it i was just asking thats all thank you so much for the update the foxy skin is great

maelstrom2001, 2017-05-08:
@dannysmobiles no problem at all :smile: it is one of frequently asked questions, we'll publish full list at soon.

dannysmobiles, 2017-05-08:
thank you is got battery v on bottom left does that read just one battery or does it read both but reads the lowers one (if you no what i mean) bar from that im glad is come out for the predator :)

maelstrom2001, 2017-05-08:
@dannysmobiles it's the lowest battery voltage. You can see both batteries voltages on info screen, or configure one info line to display them consequentially.

dannysmobiles, 2017-05-08:
thank you so much for your time and keep up the great work ;)

closed 2017-05-09 by TBXin