maelstrom2001/ArcticFox #330: Opened firmware is not suitable for the connected device

Gristrosia, 2017-05-08:
Hi,

I'm getting this message (_Opened firmware is not suitable for the connected device_) when I try to update my VTwo mini with the latest firmware and NF Toolbox 2.0. I found two older bug report with similar issue. 

I reverted to the stock firmware, I can update to previous firmware but still cannot upgrade to the new af_170508.bin firmware. 

Any idea? Is it "safe" to force the firmware update using the CTRL+ALT combo?

Thanks.

maelstrom2001, 2017-05-08:
Hello! Re-download binaries, maybe something goes wrong. On my side:

Gristrosia, 2017-05-08:
I tried to re-download the binaries, same issue. 
[image]

I'll try tomorrow on another OS like Windows 7.

Gristrosia, 2017-05-08:
SAme issue with my Primo mini :(

[image]

nifelvind, 2017-05-08:
Maybe try another usb port

TBXin, 2017-05-08:
@Gristrosia try to redownload firmware file via another browser. Seems your browser or proxy corrupted file during download process.

Gristrosia, 2017-05-09:
Not sure if it's on my side. I tried all the firmwares, and I can update up to the  af_170331 without any issues. I tried to downloaded the file several times, AV off different browser. I downloaded NF Toolbox that apparently bumped version overnight (7.07 -- I had 7.02), still getting the same pop-up. 

I tried to force flash the firmware, getting a black screen on my evic VTwo mini.

Name:	af_170508.bin
Date:	5/9/2017
Size:	112 KB (114,688 bytes)
--------
SHA-1:	802bf0357899720f2daf9ddb2850f94cb06991e7
MD5:	593e4ca7152ea60e5764cef46d796634

Extermicide, 2017-05-09:
Had same problem... Realized I was trying to flash the new binary with the old version of "NToolBox".

Downloaded the new version, all went well.

TBXin, 2017-05-09:
> Not sure if it's on my side. 

I'm 100% sure, because many people have already written about the successful installation.

> I can update up to the af_170331 without any issues. 

Because you are running old NToolbox version. Close all instances and then start new one from 7.0.7 folder.
Your welcome screen should look like this:

[image]

> re-flashed my original Wismec binary using their software.

This is not necessary operation.

>  Downloaded the new "NToolBox", and flashed the new ArcticFox binary.

Yep, this one helps ^^

Gristrosia, 2017-05-09:
I'm using, and I used, the latest toolbox available (version 2.0 build 7.07 at the moment of writing) to flash the firmware - even the older. The one with the interface that look like this:
https://camo.githubusercontent.com/e8f75ec81e472aaad0912206d818f09906e52885/687474703a2f2f692e696d6775722e636f6d2f6675784939746c2e706e67

Still getting 
[image]

GnFlala, 2017-05-09:
had exactly the same problem and all i did was to close Ntoolbox. Then i opened NFirmwareEditor (not ntoolbox) directly from the folder, pressed on Tools->Firmware Updater->From file.
That way i managed to update to latest firmware. 

From within Ntoolbox Firmware update i was getting Not suitable for this device. 

I don't know if that helps, but its all i got.

ReMicroN, 2017-05-09:
Maybe another HW version?

TBXin, 2017-05-09:
@Gristrosia redownload NToolbox archive and extract it to the newly created folder, eg: C:\NFE\
Also you can try to start NToolbox.exe with option "Run as administrator".

khamien, 2017-05-09:
Same here running windows 7. Tried multiple USB cables on different USB ports. Tried administrator, NFirmware editor. Same error message on RX200, RX200S, PICO and PREDATOR. The 3 first ones accept 170201, but no way to push 170508.

Gristrosia, 2017-05-09:
Thanks GnFlala, I managed to flash my evic vtwo with the latest firmware.

Starting NFirmwareEditor.exe crashed on start with the error log complaining about the implantation of FIPS. I disabled FIPS in Windows Local Security Policy, tried to flash, and it worked. 

If anyone has the same issue trying to "af_170508.bin" upgrade to check if "_Use FIPS 140 compliant cryptographic algorithms, including encryption, hashing and signing algorithms_" is disabled. "Disabled" is the default settings, but some Wifi drivers enable the setting when you install them. Or if you followed a STIG to secure your machine ...and forgot about this problematic setting like me;;

TBXin, 2017-05-09:
@Gristrosia thanks for the explanation!

A little excerpt from the article:
“FIPS mode” doesn’t make Windows more secure. It just blocks access to newer cryptography schemes that haven’t been FIPS-validated. That means it won’t be able to **use new encryption schemes**, or faster ways of using the same encryption schemes. In other words, it makes your computer slower, less functional, and arguably less secure.

That's why NToolbox can't decrypt firmware and find related PID.

Thanks again.

khamien, 2017-05-09:
[image]
FIPS already disabled here. Will try on another computer when I have a chance.

Gristrosia, 2017-05-10:
@TBXin yes, I usually disable it on my home desktop, but keep it enabled on my laptop where FIPS being enabled is more critical/important.  I have the same GPO backed up for both machine, and I forgot about it/to turn it off on the desktop when I upgraded to Windows 10 rs2 :/ 

If you want more detail about why or _not_ enable it and 
https://blogs.technet.microsoft.com/secguide/2014/04/07/why-were-not-recommending-fips-mode-anymore/

khamien, 2017-05-10:
Gave up trying on my own computer, no idea why it does not work while precedent releases did.
but I could manage to flash my predator today from a computer at work -really appreciated NFtoolbox beeing a no-install folder- also running windows7. 
You guys rock, after using "myevic" I loved ArcticFox on my other boxes, and really expected it on wismec predator, relieved. Thanks so much for bringing that to us, congrats and maximum support!

forkoz, 2017-05-11:
Had this issue but updating toolbox allowed the upload.

closed 2017-05-16 by TBXin