maelstrom2001/ArcticFox #335: Predator 228 not lazy enough?

joeblowma, 2017-05-09:
Seems when I switch batteries, or power off then on, it's still doing the profile detection - it shows the profile I had set before turning it off for 1s or so, then changes. Using 170508, of course. Tried resetting dataflash, then set up profile1 as TC, and then profile7 as VW for the same coil, never removing it, and it recurs.

I think the VTwo did it a couple times, but that could be my imagination as I've not been able to get it to do it again, and pico I've had no such issue as of now. Not sure if any others besides these have an issue, turned off smart entirely (for the time being, I hope! ^^).

aside: great update, thanks yet again for doing this firmware!

TBXin, 2017-05-09:
Hello,

Lazy Smart can't handle case when you de-energize the board. 
For firmware this situation are equals to the OS startup, but with out any knowledge about previous state.

joeblowma, 2017-05-09:
Hrm... it has the right profile shown on power up whether I remove the batteries or just shut it off via clicks, it just changes it after 1s to the first that matches the set resistance within the threshold like 'smart' instead of 'lazy'.

Plus, for whatever reason, it seems only predator that does it... I just left the battery completely out of the pico for 5 min and it kept the profile I had selected (7 vw vs 2 being the tc for the ato). I know the vtwo has that little tiny "battery" in it to keep the clock active during battery switches so I'm not trying too hard there. I'll leave the battery out of both for 6+h and see if that makes a difference, but I doubt it will.

I've seen sometimes/rarely after leaving the mods for an hour or more after turning off, when I turn on, if I try to fire too soon it gives me a no atomizer error - maybe this is more the case immediately with predator and AF is detecting that as an 'active atomizer change'? If so would it not make more sense to start seeking from the current profile instead of always from the first one, at least in 'lazy'?

Anyway, I can live without lazy smart, I'm pretty happy with it even if I have to press more buttons sometimes! Soooo much better than OEM.

^^
> Off/On/Lazy - Smart Mode behavior setting, Lazy means that profile can be switched automatically only on atomizer change while the box is in active state  _, when you turn it off then on... sometimes, or change batteries... sometimes_;

Got it!

closed 2017-05-09 by TBXin

joeblowma, 2017-05-09:
Not sure it matters at this point, but I left the battery outside of vtwo and pico for around 10h and neither of them switched profiles. Predator, every time I shut it off without taking battery out, or take the battery out.