maelstrom2001/ArcticFox #336: Logo unable to upload from wismec updater/ not showing on main screen

STRETCHDEATH, 2017-05-09:
not sure whether there is no logo uploaded currently, or how to upload a logo through nfirmware seeing as wismec updater does not recognise the build to be able to upload that way. any advice is appreciated other then this the new build is amazing on my rx300!

joeblowma, 2017-05-09:
If you open NFirmwareEditor.exe from the NFE package, go to 'Tools' menu and select 'Firmware Updater' there should be a button there that says 'Logo' on the bottom left. As far as I can tell, there is no logo for the logo options until you upload one that way.

ETPNV13, 2017-05-09:
I could never get nfirmware editor's logo updater to just load my logo without "converting" it, so I just use "vtc font" firmware updater to load my logo's since I switched over to arctic fox and it has always worked for me, you can get it here... https://www.dropbox.com/s/83zd19gu05...CFont.rar?dl=1

TBXin, 2017-05-09:
> I could never get nfirmware editor's logo updater to just load my logo without "converting" it,

Opion 1:

[image]

Option 2:

Select 1bpp bitmap files to upload.

ETPNV13, 2017-05-09:
Do I leave the "threshold" at "127" to keep it "as is"?

TBXin, 2017-05-09:
@ETPNV13 it is a default value for algorithm, but you can change it and see what you get right after value change.

ETPNV13, 2017-05-09:
@TBXin Thanks! This will make it much easier when flashing over all my devices to new firmware, it was rather annoying having to swap back and forth between programs to do it before lol

TBXin, 2017-05-09:
@ETPNV13 Hi again :)

Added new tab "Screen -> Logo" to the ArcticFox Configuration. Now it is possible to upload or remove logo in the ArcticFox firmware via NToolbox ^^

You can test nightly build from here if you want ;)

https://ci.appveyor.com/project/TBXin/nfirmwareeditor/build/7.0.14/artifacts

closed 2017-05-10 by TBXin