maelstrom2001/ArcticFox #337: F'n love it! Found a couple bugs though and have a couple requests.

DamnMyEyes, 2017-05-09:
First off, thank you so much!  My assumed piece of shit Predator 228 is no longer shitty, now that TC works because of AF. This is my first time using AF and all of the features and options are really amazing.

The bugs I found:
1.) If the device is locked and keys are not locked, if you hold down the -/+ they will change the temp/watts.

2.) This might not be a bug and is probably more of a NFE Toolbox issue.  While using the device monitor, the Power line will not exceed the Power Set even though I have a boosting preheat set (Relative 120% for 1sec).  If I hold the fire button for less than the 1sec preheat duration, my screen will show the Last Power was in fact higher than the Power Set.

My requests:
1.)For me and my fellow products of the US public school system, it would be nice to be able to change the date and time formats to the backwards way we are used to.  12h time format (AM/PM probably isn't necessary and a waste of space) and MM/DD/YYYY

2.)It would be cool to have the option to chose what is displayed at the top left corner, such as puff counter or time rather than the Preheat, Stealth Mode and LSL indicators.

Thanks again and keep up the great work!

TBXin, 2017-05-09:
Hello,

Thanks for your report.

> If the device is locked and keys are not locked, if you hold down the -/+ they will change the temp/watts.

Yep, this is a little bug ^^ We will fix it as soon as possible.

> If I hold the fire button for less than the 1sec preheat duration, my screen will show the Last Power was in fact higher than the Power Set.

This is by design. If you are pressing fire button less than preheat duration - preheat will be enabled on the next press too. This is a "protection from misscliks".

> MM/DD/YYYY

We will think about this possibility.

> It would be cool to have the option to chose what is displayed at the top left corner, such as puff counter or time rather than the Preheat, Stealth Mode and LSL indicators.

Oh, UI is a so personal thing... We have an couple of ideas for the future releases... I can't tell now, want that it will be a surprise ^^

Poopcheeks, 2017-05-09:
i argee on the date and time haha its very confusing for US in the states

GnFlala, 2017-05-09:
all i can say about the new skin is congrats. You really nailed it. All the info you need on the screen. No need to go into the menu to look for anything. Most handy thing is the locked resistance. Now i will never mistaken have higher one.

Only suggestion would be to have the battery percentage to the bottom right of the screen, where all battery info are (battery voltage, battery bar), and board temperature to go to the upper right corner.

 On the upper left corner where the dots are (they get replaced by P, S, L if we enable Preheat, Stealth or LSL) I would prefer icons instead of letters. Lets say Fire icon for preheat, a stealth icon like this https://www.shareicon.net/interface-ui-spy-multimedia-option-icognito-851310 for stealth and something for LSM. Icons help to understand better what they mean.

Anyway congrats to the devs and thanks.

maelstrom2001, 2017-05-09:
> If the device is locked and keys are not locked, if you hold down the -/+ they will change the temp/watts.

[Fixed](https://github.com/maelstrom2001/ArcticFox/releases/tag/170509) :smile:

TBXin, 2017-05-09:
> I would prefer icons instead of letters

You try to draw icons 8x8 readable icons ^^

GnFlala, 2017-05-09:
The website i posted goes down to 12x12 . If the paint is simple it is doable.

GnFlala, 2017-05-09:
Example. And they look very good. 

http://wiki.baltimorenode.org/index.php?title=File:8x8_icons.png

javak810i, 2017-05-09:
I also have a small design change suggestion. With that it would be perfect (at least to have the option). I'm not a big fan of the new non-removable battery meter at the bottom. [image]

DamnMyEyes, 2017-05-10:
If the device is locked and keys are not locked, if you **hold** down the - **or** + the temp or watts will change.

> Fixed 😄

Sorry @maelstrom2001, but this doesn't appear to be fixed in **[170509](https://github.com/maelstrom2001/ArcticFox/releases/tag/170509)**

The date and time format does appear to be changed though!  Was someone sneaky and slip an undocumented fix in?

TBXin, 2017-05-10:
> The date and time format does appear to be changed though!

And should not have been.

forkoz, 2017-05-11:
Yes please, let us change the month for day. The bigger clock is better than AM/PM though.