maelstrom2001/ArcticFox #340: Predator 228 atomizer short

gray584321, 2017-05-09:
My predator with af keeps giving me an atomizer short error with all of my atomizers. I flashed it with the stock firmware and everything works fine. Any advice?

maelstrom2001, 2017-05-09:
Seems the same as #323. 510 grounding is the sore spot of new boxes.

ETPNV13, 2017-05-09:
If you have any experience with soldering, just solder it to its base so it will not pull out or throw an "atomizer short"

escoltajuverf, 2017-05-10:
same issue with me predator and rx 2 /3 atomizer short but works fine with stock firmware, work around is reflash firmware->erase dataflash and dont ever open powerbank mode again, also charging via usb triggers the issue, the atomizer didn't short the resistance didn't change, something just trigger the atomizer short message

ETPNV13, 2017-05-10:
None of these issues have happened to my predator mods and I have 3 of them, I charge occasionally via USB, I have put them into powerbank mode and I have never had to erase my dataflash or reflash my firmware other than when I make tweeks via nfirmware editor to make things to my liking, I did get the dreaded "atomizer short" error but after soldering the 510 threads to the casing I have had no further issues and that only happened on one of them so far

ETPNV13, 2017-05-10:
To be clear, I had the atomizer short on stock firmware when the 510 threads started coming loose from their press fit and it was soldered before this version of arctic fox even came out but for 2 days I have been playing with all the features and have not encountered a single issue as of yet on any of my 3 predator mods...

andreivoinea93, 2017-05-10:
How did you solder the 510 ground? On Evic Primo it's made of aluminium I believe. Solder just won't stick to it.

ETPNV13, 2017-05-10:
I soldered it from the bottom side and sanded and applied flux prior to soldering so maybe that's what made it "stick" I haven't had any issues since and haven't checked it since but my understanding is that it may cause corrosion problems in the future from the differing metal compositions, I also use heavy lead/light tin solder so I dunno if that makes a difference

ETPNV13, 2017-05-10:
I attempted just soldering the top first but that didn't "stick" lol

TheSpiritof69, 2017-05-10:
@ETPNV13 If it's not too much effort, do you mind sharing a picture?

I had mine taken apart the other day to fix the sticky firebutton (juice gunks it up really nicely) and am curious how you soldered it in case my 510 pops out in the future *knock on wood*.

FWIW, for the firebutton I considered a few dabs of silicone around its case to keep juice from seeping in..might do that if I have to open it again for that reason..

ETPNV13, 2017-05-10:
It is a HUGE pain in the ass to do this, you have to sand the metal first (very hard in tight space you have) and then apply soldering flux and hold the iron on the two metals for like two minutes just to get the metal casing hot enough to "take" the solder but here is pics..
[image]
[image]

ETPNV13, 2017-05-10:
My soldering iron may just need cleaning and is getting old but it took me forever...I can't really suggest anyone do this unless you know what you are doing as it is not easy

ETPNV13, 2017-05-10:
Alternatively, I think super glue on just one side (to maintain conductivity on the other side) would have been easier...just a thought

andreivoinea93, 2017-05-10:
no matter how close you try to glue them, there will be an air gap between the wire and the connector without soldering...so yeah, dont think it's a great idea

TheSpiritof69, 2017-05-10:
@ETPNV13 Thanks for the pictures. I agree that its likely a pain in the butt to do, but what can you do...

I guess its the only viable option to fix Wismec's shit permanently in this case.

ETPNV13, 2017-05-10:
Personally, I never had any issues with the previous rx200 style 510's, I use multiple different styles of tank that all screw all the way down without issue (including the smok tfv8 that everyone said would "destroy" the 510 connection) and I don't understand why they changed it other than a million complaints about how "horrible" they were. Neither I nor any of my friends ever experienced these problems so maybe it just depended on the batch...but I will say that fit and finish of joyetech/eleaf/wismec is definetly hit and miss as I have 7 rx200s mods and only 3 of them "fit" right with the battery doors and front panels...but that is just being nitpicky about the machining of the individual parts, the boards have never failed me and neither did the old style 510's

oddkode, 2017-05-16:
I've been having similar issues, except my 510 seems to be sturdy. I get the atomizer short message when I change tanks or randomly.

I've been trying something, though - but I can only confirm it works in VW / power mode. From main settings (fire + wattage up), highlight your active profile (I named p1 "power" for vw), hit fire, on the next menu go to coil, hit fire, then manually enter your ohms. It doesn't need to be bang on but try to be close. Then click fire to save it, and exit the menu.

You may need to do this a few times but I find once you get it and it stops giving you the short message, it'll stay stable for a while at least. Could that be a bug (or if it's still a hardware issue, maybe it's a helpful feature, heh)?

Oh and conductive epoxy would be another way to fix the press fit 510 negative piece (maybe after giving both the 510 piece and the inside of the body connection a bit of a light sanding), apply a little to close the gap and that should work.

Problem is silver epoxy will run you a pretty penny. A kit of 0.2gm syringes costs almost $60 in Canada. More than the price of the mod :( Not sure if cheaper conductive epoxies like carbon / graphite ones would work, may not due to the resistance differences.

cdyseplc, 2017-05-16:
is there a tutorial or sth for completely dismantling the predator 228 that anyone an point me out to? i got so far as removing the blank panel and the battery compartment, im looking to clean up my firing button internals and maybe glue down my 510

joeblowma, 2017-05-17:
@cdyseplc Not the best quality video, but [this one](https://www.youtube.com/watch?v=PQ-AgB1Bm7g) shows it all.

cdyseplc, 2017-05-20:
thank you so much

tintreach, 2017-05-21:
Mine started doing this and I noticed it after the latest firmware update.  510 is still rock solid, no wiggle at all.  Reattaching the atomizer resolves the issue.  Issue only happens when it comes back from a sleep/standby mode.  Normal vaping, no issue, goes to sleep and wake it back up... atomizer short.