maelstrom2001/ArcticFox #343: not a issues/ just a thanks for your work

docdeath, 2017-05-09:
big thanks to team here for wicked firmware and updates for rx300 /and other mods to

acevalsl, 2017-05-09:
Same.....txs

kyan31, 2017-05-10:
Yep, thanks to all of the Arctic fox team for developing such a wonderful firmware. I just found out about this today and I'm loving it, literally made my RX2/3 a whole new device. It's just like having a DNA board, I'm actually lost for words because I didn't think the standard chips were capable of all these features. Thank you again!

snigel, 2017-05-10:
Great stuff!

acevalsl, 2017-05-10:
Where can i make a donation for your job??

TBXin, 2017-05-10:
> Where can i make a donation for your job??

Here https://nfeteam.org you can find donate button on the top panel.

acevalsl, 2017-05-10:
donation PayPal made
txs

TBXin, 2017-05-10:
> donation PayPal made

Thanks! We really appreciate it.

derTim, 2017-05-10:
Also big thanks from me! Really missed AF on the new Primo 2.0 :)

amrdia, 2017-05-10:
u guys are awesome

Sonicaki, 2017-05-11:
I would like to say a big THANK YOU too!!!
Foxy skin rocks!! perfect design!
Thank you guys for everything!

Hunterror77, 2017-05-12:
Thank you all , keep up this excellent work!

closed 2017-05-15 by TBXin