maelstrom2001/ArcticFox #344: Puff Display Gone / Small Screen

messywrk, 2017-05-10:
Eleaf Pico / Small Screen

In the newest versions the puff bar gauge is gone. Previous there was a graphic indicator of the puff together with the current puff length.. Now the screen doesn't change while firing and stays at the default layout.

Can we please get this feature back?

maelstrom2001, 2017-05-10:
> Previous there was a graphic indicator of the puff together with the current puff length.

There was stock firmware. Puff bar was never implemented in ArcticFox.

[image]

messywrk, 2017-05-10:
Damn, you're right. Sorry! 😁 

It's just the change in Layout if you enable "1 Watt Mode", that threw me off..
Before 75.0W -> now: 75W

In the last versions you had 1 digit precision no matter what you selected. I guess it's just a matter of preference, I always like the more techy look. The screen space shouldn't be an issue as long it's a <100W device.

Feel free to change this back, If I were you, I probably would not because of the >100W devices.. ;)

ArionWT, 2017-05-10:
>The screen space shouldn't be an issue as long it's a <100W device.

I completely disagree with this point. If "1 Watt per step" is enable, ".0" is just a dead weight. I think, stuffing free screen space with useless info isn't the best idea.

closed 2017-05-14 by TBXin