maelstrom2001/ArcticFox #346: HW Version is overwritten

aaf-caliban, 2017-05-10:
Found only with iStick Pico Mega
the HW Ver says 42949670.00
The Updater says Hardware Version -0.01
I used NToolbox 7.07 and 5.09
One can correct this in updater, but in my case, i couldn't remember my original version ;-)
But there seems to be only one hardware version of the Pico, so this was no problem. I was not able to reproduce this issue.

TBXin, 2017-05-10:
Hello,

If with -0.01 HW Version your pico still works fine thats means that your HW version was 1.00.
At this point we know about only two HW versions for pico: 1.00 and 1.01 and 1.01 uses another display, which works a bit different :)

aaf-caliban, 2017-05-10:
Thank you for the clarification. I set it to 1.00 and it works :-)

closed 2017-05-10 by TBXin