maelstrom2001/ArcticFox #347: Battery Profiles CSV import?

IngoPOW, 2017-05-10:
Hi,

what about a new function in the future to import battery csv´s?
So users are able to import DNA battery configs that are well known and tested out.
I haven't found no battery capacity files in .xml yet.

regards,
ingo

TBXin, 2017-05-10:
Hello,

DNA's discharge profile are pretty different from ArcticFox discharge profile.
Automatic conversion can't be made.

Anyway, you can open CSV file via your favorite text editor and transfer values manually to the Discharge Editor Window.

IngoPOW, 2017-05-10:
yeah, thats the way I do it atm.

hawkeyexp, 2017-05-11:
I have done it same way and manually set the rounded values from dna profile. But would be great if import can handle this.

TBXin, 2017-05-11:
@hawkeyexp DNA's discharge profile can contains variable count of the points while ArcticFox discharge profile always contains constant number of point.

Interpolation is not a solution, since discharge curve is not linear.

At moment i don't know how to handle all possible cases to create AF profile from DNA profile.
This looks like a complex task with no profit for most users since they are using GEN profile.

IngoPOW, 2017-05-12:
well i tested out the gen profile and put manually the dna profile data from vtc5 into the editor.
as a test result the battery icon was more precise with the dna data.
i will do time measurement the next days to see if it was only felt longer or realy lasts longer.
for a startup it would be cool to have a .csv import filter added instead of the .xml filter.