maelstrom2001/ArcticFox #350: ELeaf iStick QC Atomizer short

adzam016, 2017-05-10:
Hi Guys,

Been running the new firmware on my iStick QC and I keep getting atomizer short.

If I take the atty off and on again, all is fine.

No liquid on contacts whatsoever and using Smok Big Baby Beast so leaking is albeit if ever.

My iPower is absolutely fine on the new firmware however.

The atomizer short issue does not arise with stock firmware, in fact this month old mod has never had any errors until now.

Sure it is just a little bug in the firmware and I will keep it on there to see if there any more bugs as this mod is new to your firmware.

Thanks,

Adam

TBXin, 2017-05-10:
Hello,

This problem occurs only on the few box models and i think they all have similar construction. 

You can read more in the related issues: https://github.com/maelstrom2001/ArcticFox/issues/340 https://github.com/maelstrom2001/ArcticFox/issues/323

andreivoinea93, 2017-05-10:
@TBXin I know this is just covering the dirty hardware design, but shouldn't there be an option available to lower the overcurrent threshold so we don't have this problem, like it is with the original firmware?

Again, I understood the problem and I am aware it's not software related, actually the software is too good for some of the hardware models. This would just cover the problem, not fix it, but still, it sucks not being able to use AF :(