maelstrom2001/ArcticFox #352: RX200s - 2 Batteries

billkabb, 2017-05-10:
Hello 
with firmware 170212 i was able to use my RX200S (@max150w - 6 volts) with only 2 batteries using a diy passthrough on the right (3rd) battery place  .  with the last update 170510 it doesnt work and it says check battery .
is there a way to bypass this check ? 
or maybe to force the installation of RX2/3 firmware ?

billkabb, 2017-05-11:
https://goo.gl/photos/9ZNzn9KUyC1gsfPj6

maelstrom2001, 2017-05-11:
Try attached build. From some time 2 of 3 cells configuration was limited to RX2/3 only.
[af_170511.zip](https://github.com/maelstrom2001/ArcticFox/files/993849/af_170511.zip)

billkabb, 2017-05-11:
Yes ... 170511 firmware works great with 2 batteries... thank you sooo much ... 
greetings from greece ...

closed 2017-05-11 by billkabb