maelstrom2001/ArcticFox #354: ohm value decrease when firing?

edward81, 2017-05-10:
Evic primo with dual coil ss316l.
I just updated the new firmware (can't confirm with the old) 170510. And i get this strange behaviour. The resistance decrease when i fire. After firing the resistance go up again.
[image]

maelstrom2001, 2017-05-10:
It is normal behavior. In standby mode all boxes shows increased Live Resistance. Some higher, some lower, but all of them.

cpulley, 2017-05-11:
Is it possible to add an offset for this, maybe an expert setting?  I'm not 100% sure if it's the same issue, but my Cuboid is showing a coil at 3.13 ohms at rest, when it's actually a 1.05 ohm coil. Sadly, this makes most of the info on the screen useless, and limits me to 25.9W (as it thinks it'll be firing at 9V)

[image]

Justin-L-Franks, 2017-05-11:
@cpulley I had a similar issue recently on my Cuboid running the myevic firmware -- a 1.2 ohm coil was reading as 2.1 ohms cold, but when firing read correctly. Turns out the 510 connector was just a bit dirty. After cleaning it with a bit of alcohol and cotton, the problem disappeared.

edward81, 2017-05-11:
How i can lock cold resistance if the mod show me the wrong value?

TBXin, 2017-05-11:
> How i can lock cold resistance if the mod show me the wrong value?

You can lock any resistance and then correct value in the Profile Menu of the device or in the NToobox.

closed 2017-05-16 by TBXin