maelstrom2001/ArcticFox #355: Possible safety issue when 510 connector is shorted

Justin-L-Franks, 2017-05-11:
I was testing how much travel the spring-loaded positive 510 pin had on my new eVic Primo 2.0 with a metal screwdriver and noticed a clicking sound when the screwdriver came into simultaneous contact with the threads. It pulses at around 4 Hz, and sounds like power is being applied across the connection. Once the mod goes into sleep mode this behavior stops.

(Build 170510)

joeblowma, 2017-05-11:
You've pretty much described how a digital multimeter measures resistance, passing a low known current across the resistance and measuring the resulting voltage. Completing the mod's circuit with a screwdriver should net a 0 resistance, once it pulses to check the resistance. Also, some of these mods are known to make a clicking noise when they sample the resistance when idle, or even while firing as they tend to sample then as well.

Justin-L-Franks, 2017-05-11:
So this is normal? The clicking sound is pretty loud, and it sounds like a fairly significant amount of current is being passed. I even saw a tiny spark once when contact was made.

TBXin, 2017-05-11:
All devices make clicking sounds if you bring them to your ear, even DNA based devices.
Someone louder, someone not. Also loud is depending on the resistance of the coil.

Justin-L-Franks, 2017-05-14:
The reason I'm concerned is because it clicks even when not firing, and does not stop. After a couple of seconds at 4 Hz it drops down to 2 Hz, but continues as long as the connection remains shorted.