maelstrom2001/ArcticFox #356: Suggestions for Foxy skin

Justin-L-Franks, 2017-05-11:
My new eVic Primo 2.0 was delivered today, and after a quick function check with the stock firmware (to be sure everything was working properly), I installed ArcticFox build 170510. The new 'Foxy' skin is **very** impressive -- much more information is immediately available yet everything is still easy to read, and the overall look is quite clean and elegant. You guys really did an amazing job with the new skin.

With that said, I do have a few suggestions for further customization:

- There are three separate battery displays (percentage at the top left, voltage at the bottom left, and a segmented bar at the very bottom. We should be able to edit what information is displayed in those locations. I would love to have a small digital clock (12-hour HH:MM with AM / PM) in the upper right where the battery percentage is displayed. I'd also like the board temperature at the lower right replaced with the battery voltage of the second cell (so it would show the individual voltages for each cell. And I would also like to be able to switch the segmented battery bar at the bottom to a puff duration bar (could be either segmented or smoothly lengthening), with the option to have it reset to zero after the puff, or remain at the length of the puff (this would give an estimate of the last puff time, giving you an extra bit of information displayed in the currently-editable three lines if last puff time is something you want displayed).
- The three currently-editable lines show their values in reverse (black on white instead of white on black). I would like to see an option to change this to white on black.
- Digital clock should be selectable between HH:MM:SS and HH:MM, in both 24-hour format and 12-hour format with AM / PM.
- A fourth line of data should be able to be added with an appropriate decrease in size of the wattage or temperature display. The wattage/temperature display would still be plenty large enough, and this would allow for a fourth line of data for those who want it.

I believe these features would add a new level of customization to an already great skin.

b0taSs, 2017-05-11:
Just to add on this, 

- The current display on the foxy theme seems to be related to the atomizer current, and unlike the circle theme (idk about the regular, never used it much c: ), we lost the battery current which was nice to check headroom.

TBXin, 2017-05-11:
> we lost the battery current which was nice to check headroom.

ArcticFox firmware never showed and will not show battery current because PCB can't measure it.

b0taSs, 2017-05-11:
@TBXin Well, I went to check and you're right 😅 

I made that assumption because of some weird readings on the "Output Amps" when I updated the firmware the last time, but I can't seem to reproduce them so, yeah, my mistake 😄sorry about that

TBXin, 2017-05-11:
@b0taSs no problem 😉

sarathmenon, 2017-05-14:
Can I tack on a feature request to this? On the classic skin, my preffered setting for battery display is +Volt(s) to see the voltage of each individual batteries. I am familiar with LiPo batteries from my other toys and this is for some reason the most intuitive setting. I lost that ability with the foxy skin, I can only see an average Voltage at the bottom. I feel more comfortable seeing the individual levels. Is it possible to remove the battery indicator and the percentage at the top and make room at the bottom for both the battery levels?

Foxy is a great effort, and miles above the official Joyetech screen, but sometimes too much information can be a thing. Thanks for all your work.

maelstrom2001, 2017-05-14:
@sarathmenon https://github.com/maelstrom2001/ArcticFox/issues/372#issuecomment-301298396

sarathmenon, 2017-05-14:
Thanks, I didn't realize those dots were for the different settings. I got that scrolling battery level to work when I tried it, but for some reason I like the Classic Skin display where both the battery levels are visible. It's a minor nit pick, since I don't see how you can display both levels with that minimal screen.