maelstrom2001/ArcticFox #358: Logo function not working and no games/3d screensaver on RX2/3

kyan31, 2017-05-11:
On my RX2/3 I have build 17509 and my logo function doesn't work. If I apply it on the main screen and set an interval, it just never shows up. If I try putting a logo on the EXT charging screen it shows the normal charging details but where the logo would be it's just black. Also another small issue, I don't seem to have any options for games/3D screensavers like I've seen on other peoples devices. Did it get removed in the update or something? By the way these are all small issues, so please don't take it too seriously, I just want to know if other people have this issue or not so I know if it's just me installing the wrong firmware or something. Besides these little things the whole firmware it's working great and I'm so thankful for it, turned my standard RX2/3 into a DNA device, maybe even better.

m600x, 2017-05-11:
As far as I know, Games and 3D are on the myevic firmware (outdated) and not ArcticFox which highly seems like a fork of it with closed source.

kyan31, 2017-05-11:
@m600x the thing is I have seen people with the RX2/3 on YouTube who have logos, games, and the 3D all working. Watch this video https://youtu.be/8ZLvJ9EP5bs

Skip to 4:18

m600x, 2017-05-11:
... Did you actually read what I said? The relevant part is : "Games and 3D are on the myevic firmware (outdated) and not ArcticFox"

This mean the firmware you see in your video is NOT ArcticFox but myEvic. He also point out that he made an updated version of the video where he install the ArcticFox firmware.

kyan31, 2017-05-11:
@m600x ah I understand now, my bad on that one. I didn't know what the myevic firmware was, I assumed it was Arctic fox he was running just from watching the video, really should have read the description. Anyway thank you for the quick replies I appreciate that 👍👍

closed 2017-05-11 by kyan31