maelstrom2001/ArcticFox #361: Battery cut out at 3.5V

yoshy262, 2017-05-11:
Please, can you set the batery cutout at 3.15 V or 3.25 V?

The Wismec Fw cutout the batery at 3.25V, Arctic cutout at 3.5V.
Is not quite optimal for a batery to be recharged at 3.5V because is not drained.
I know, is a safety issue but 3.15 or 3.25 is very, very safe to drain a battery till at 2.9V or 3V.

TBXin, 2017-05-11:
Hello,

This was a little bug in the GEN discharge profile.
You can install latest available release https://github.com/maelstrom2001/ArcticFox/releases/tag/170510
And problem should gone.

Also, you can create custom discharge curve with custom cut-off value and custom points.

yoshy262, 2017-05-11:
Thank you very much for a very fast answer.
I already put the new Fw on the Predator.

closed 2017-05-11 by TBXin