maelstrom2001/ArcticFox #362: Small Screen

UsernamesAreAlreadyTakenn, 2017-05-11:
Hello,
I'm searching for the "small screen option", I know she exists but I don't know where, could you tell me please (fool question)
I can't find it in the section "appearance"

Btw thanks to devs, you did an awsome job !

TBXin, 2017-05-11:
Hello,

Small screen skin applying automatically if you are using devices with a small screen.
You can change layout here: Screen -> Layout -> Small Screen

If you want to choose skin for small screen devices you should look here: Screen -> Appearance -> Main Screen Skin: Classic / Lite.

All this options are available only in the NToolbox.

UsernamesAreAlreadyTakenn, 2017-05-11:
I'm on RX300
I'm searching for an option that disable the profile name in the circle skin
I tried to write nothing in the name section, but there is still a free space, taken by nothing

TBXin, 2017-05-11:
RX300 is a "big screen" device.
There are no such option to remove profile name from a screen.

UsernamesAreAlreadyTakenn, 2017-05-11:
Ho ok thanks
Can I ask to you to put an option which remove this free space ?
(But I think that is probably too complicated for nothing, and I can understand)

TBXin, 2017-05-11:
We can't put an option for every UI change request from users. This is a matter of taste for every person. 
One person want to remove free space, second want to change font size, third want to put more info and so on.

The first reason is that we don't have unlimited space for firmware in the device.
The second reason as i wrote above is a matter of taste of each person. The number of settings will tend to infinity.

Maybe in the foreseeable future we will give users some more flexible configuration tool, but not now.

I hope you understand us and do not hold resentment.

UsernamesAreAlreadyTakenn, 2017-05-11:
No problems, I can perfectly understand that. Thanks for quick response :)

closed 2017-05-11 by UsernamesAreAlreadyTakenn