maelstrom2001/ArcticFox #363: Predator 228 battery life

DamnMyEyes, 2017-05-11:
I apologize if AF has nothing to do with this, but I have never used my Predator (or any Wismec) with factory firmware for more than a few minutes. A google search for 'Predator 228 battery life' didn't find any answers.

My batteries (LG HG2) are lasting about 5hrs in the Predator 228 w/ AF 170510.  For comparison, my Smok G-priv lasts at least 15hrs using the same batteries, same atomizers (Troll RTA, Cleito 120, Moonshot 24 v2) and very similar preheat/TC settings (SS316L @ 450°F with 75w and ~90w preheat for 1sec).

I saw there was a generic battery profile change in the last nightly (170510), so I installed it hoping it might fix my issue.  I don't notice any difference between 170510 and the previous 170509 nightly build.

Any help would be greatly appreciated.

Keep up the awesome work!  Overall, Arctic Fox is a dream come true!

lentis720, 2017-05-11:
Mine lasts 2 days easily vaping all day long at 20 watts. I'm using Sony VTC6 (comparable to HG2).

For comparison my RX200S which has 3 batteries lasts 3 days, so all seems fine.

I know you're using the 170510 build but check your battery voltage when it's "empty". It should be close to 3.1V.

maelstrom2001, 2017-05-11:
> It should be close to 3.1V

I think, it should be higher a little, something about 3,2..3,3 volts because of this:
> 75w and ~90w preheat

> at 20 watts

Yes, my power settings is about 11..12 W, and I can discharge 25R in Primo (which is very similar with Predator) down to "Battery Low" at 3,12..3,14 V in standby too, using custom discharge profile:

[image]

DamnMyEyes, 2017-05-11:
I'm currently at 10% showing 3.3v.

I just remembered another odd thing.  The other day my batteries were getting pretty low so I threw it on a USB charger.  It displayed that it was pulling about .5amps but over an hour it didn't raise a single percentage.  It stayed at 11% the entire time.  That was on the 170509 build.

DamnMyEyes, 2017-05-12:
I guess the lack of change of displayed % is simply due to the steepness of the discharge curve at that low voltage.  

I customized my curve a little to experiment. I also disabled the clock (because I don't want to have to reset it twice a day) and lowered my screen brightness. Perhaps these adjustments can extend my battery life a little.

xalvarion, 2017-05-12:
I'm seeing the same problem with my Predator, batteries lasted a lot longer when I was on stock waiting for this release :D I've noticed random drops in percentage, just last night when I left home my batteries were at 100%, and after a 15 minute car ride they were at 66%. Batteries went down to 0% in 1 hour, dropping in large chunks of % at a time, vaping at 110w .30ohm TFV8 tank, batteries are Sony VTC5's, everything's about 2 weeks old at this point. When I plugged the batteries in my charger, they showed ~25% of power left. Totally new to this by the way, just saying things I noticed.

hawkeyexp, 2017-05-12:
No probs in my case on vtc6 @ 60W with imported profile for vtf6 from dna chip

ETPNV13, 2017-05-12:
I dunno if it makes that much of a difference using my own custom curves but using my custom 25r curve I get the same vaping time as I did on stock, I don't use "gen" cause I can't even see what curve they went with so I use one I can control lol

DamnMyEyes, 2017-05-15:
Just a follow up.  After disabling the clock (was in LSL mode) and lowering my screen brightness my battery performance has improved greatly.  I couldn't figure out how the battery profile would affect my battery life until I realized that it didn't and that the battery profile simply affects the displayed remaining % in the UI and nothing more.  

Feel free to close this issue.

closed 2017-05-15 by maelstrom2001