maelstrom2001/ArcticFox #365: Unable to download device settings

TiborBarath, 2017-05-12:
Hi i have problem with my Eleaf iStick Power80.
I try to use ArcticFox and i get this issue.
Anybody can help me what am i doing wrong?

[image]

maelstrom2001, 2017-05-12:
Did you flash the latest firmware? Post screenshot of Firmware Updater with connected box.

TiborBarath, 2017-05-12:
[image]

maelstrom2001, 2017-05-12:
Firmware Version is stock. It is NOT ArcticFox.

TiborBarath, 2017-05-12:
Ah i need download ArcticFox firmware and then try? :)

TBXin, 2017-05-12:
Yes

maelstrom2001, 2017-05-12:
You'll not beleive...

TiborBarath, 2017-05-12:
Its working! Thank you guys!

closed 2017-05-12 by TiborBarath