maelstrom2001/ArcticFox #366: Arctic Fox Configuration Tool total puff time glitch?

danielthelad, 2017-05-12:
When editing on the Arctic Fox Configuartion Tool the puff counter can go up to 99,999 but the total puff time can only go up to a maximum of 35,999.9. Just I had a device before with stock firmware that had over 82,000 seconds. I know it's not a major fault but just thought I'd report it. By the way thankyou for the absolutely brilliant software. Great job guys.

maelstrom2001, 2017-05-12:
Yeah, it is by design: we have no enough space on display to show total puff time in format hh:mm:ss, only h:mm:ss, so we decide to limit time to 9:99:99 = 35999 sec :smile:

danielthelad, 2017-05-12:
Okay I understand, no problem. As I said it wasn't a major fault haha. Thankyou anyway for the brilliant firmware

closed 2017-05-12 by danielthelad