maelstrom2001/ArcticFox #367: Just wondering if it would be possible...

ETPNV13, 2017-05-12:
Would it be possible to get a branch off build of the original that still supports all currently supported devices but drops all "extra" features such as power bank, clock, logo and smart mode in favor of adding two more profiles for a total of 10 profiles and making a couple more customizable battery curves and dropping non-customisable wire types to make all wire types whatever type you like (like the current tfr curves) I love the firmware and enjoy using it on every joyetech/eleaf/wismec product I own, which is a lot but I feel like there are a lot of people who just will never use all the "extra stuff" and would prefer a wholly customisable setup to make to their own liking. I know rom space is limited and not sure if what I am asking could even be done but just a thought. Also, if you find no interest in doing this yourself but it can be done then what tool set do I need to do this myself? I am familiar with a lot of different instruction sets and programming techniques but I have no idea where to even start with doing this myself...also not sure if you would even want to share how to edit this firmware as it is your baby which I wholly understand...so yeah, just wondering and thank you again for amazing firmware even as it is now.

TBXin, 2017-05-13:
Hello,

Extra feature and "settings" consumes different memory areas in the ROM.
This kind of "fork" will require significant amount of work and it will be very problematic to keep both versions up to date simultaneously.

ETPNV13, 2017-05-14:
Okay, thank you for the response!

closed 2017-05-14 by TBXin