maelstrom2001/ArcticFox #370: Preheat Curve new option request

sergiofagundes, 2017-05-13:
Is it possible to include an option to Preheat Curve to randonly use the Preheat curve for every new puff  and another one to roll throught every Preheat curve for every new puff ?

TBXin, 2017-05-13:
[image]

Justin-L-Franks, 2017-05-14:
Just set up different profiles with different curves and switch between them.

sergiofagundes, 2017-05-14:
Hi thanks.
I didn't explain well. 

Let's say there's an Option to rotate throught all the preheat curves settings for every time (new puff) we press the power button. 1st time we have a Soft curve, 2nd time a Boost 1s curve, till the last Linear curve then we go back all again from soft to linear. This way every new puff will get a new preheat curve setting automatically.

Dil3mm4, 2017-05-14:
Still 
[image]