maelstrom2001/ArcticFox #371: Curve mode but only when I didn't fire since X sec

UsernamesAreAlreadyTakenn, 2017-05-13:
Is it possible to add a feature which switch automatically in curve mode when I didn't fired since X seconds ?
That would be great for curves used to heat resistance to wanted temperature (like boost 1sec).
Like:
First fire would be a boost 1 sec to heat resistances
And if I fire 5 sec after this one, I don't need this 1 sec fire boost, then it would be automatically to absolute mode.

And if I fire 1 mn later, it would switch again in 1 sec fire boost (because resistance get cold after a non-use time)

TBXin, 2017-05-13:
Hello, @UsernamesAreAlreadyTakenn 

You can set Preheat Delay up to 180 seconds (3 minutes).

[image]

UsernamesAreAlreadyTakenn, 2017-05-13:
Ok thanks

closed 2017-05-13 by UsernamesAreAlreadyTakenn