maelstrom2001/ArcticFox #372: New AF theme - Top-left dots + a few other questions

xezsw20, 2017-05-14:
Hi AF team,
Really loving the new firmware and Foxy theme - detailed and concise, brilliant!

First question:  What are the three dots top-left for?  I can't see any notes about them in the readme,  but in some of your new screenshots they display a "p" and a "c",  is it something to do with TCR or Preheat? (neither of which I use).

Second: dual battery mods (eVic Primo v1 for me) - battery voltage meter - I see it sometimes displays Batt1, then switches to Batt2 occasionally - is this on a timer?  Or does it only show the battery with lowest voltage?
I really liked being able to see both battery voltages simultaneously in the previous release,  I can bring up two battery meters but is there any way to configure them to stay with cell1 and cell2?  Meantime I can use the Info screen to see both, but it's nice on the main screen.

Any insights appreciated.  Thank you all very much for your work,  I've sent a donation previously and I shall send another one soon.  :-)

(Apologies if this isn't the appropriate place to post all this.)
Kind regards.

Justin-L-Franks, 2017-05-14:
P = Preheat
C = (Power) Curve

They will flash during the "relax" time, letting you know that they are temporarily disabled.

Not sure what the third dot represents.

ETPNV13, 2017-05-14:
I believe the battery voltage meter always defaults to whichever is lowest

TBXin, 2017-05-14:
Hello,

Battery volts above dotted line will always show lowest battery volts.
Also you can change one of the info lines and put a "volt battery" that will rotate and show you the first battery of volts than secods, than first again, etc.

Here example how this line is look:
-- 1:3.79 ----- 2:3.83 --
[image] [image]

xezsw20, 2017-05-14:
Thanks for the responses, that answers my questions.

And monitoring dual batteries - I see now the Battery volts rotates between cells, nice.
In eScribe/DNA you can configure a cell per line, eg. "Cell1 voltage,  Cell2 voltage".  This is a great feature and could be something you could think about for a future build maybe...