maelstrom2001/ArcticFox #373: screen contrast question

ETPNV13, 2017-05-14:
Will having the screen contrast at 100% lower the life of my screen? I only ask because I was reading somewhere that dead pixels are a problem on these screens after a few months to a year...I have several joyetech mods that are well over a year old now and even though I have experienced no such issues on stock firmware will having my screen contrast at 100% start reducing the life expectancy of the screen or is it okay? If it is going to reduce the life of my screens than what setting is recommended? I think I read somewhere that stock setting is about 17% but I am not sure if that is correct...I love having it at 100% for reading the screens outside but if it means that I will fry my screens sooner I would rather have it at whatever stock settings are...

ETPNV13, 2017-05-14:
Okay after some more research on the topic I guess my main issue would be "image retention" or "burn-in" where an image is "ghosted" on the screen but also concerned with image degradation...common sense would  dictate that a lower contrast setting will obviously lead to a longer lifespan but is the resulting "increase" in lifespan going to be that much longer or is it negligible? I would much much prefer seeing the screens in daylight if it means the screen will only last a few months shorter than average but if 100% contrast is going to cut my screens lifespan in half of average lifespan than that is not worth the trade off to me...just wondering if anyone knows if this is an issue yet or not...

TBXin, 2017-05-14:
Most often, people get "faded" pixels in the battery indicator area.
Because they are charging their devices via USB in the normal mode, not stealth mode. Screens are active all charging time. This leads to a **_long operation of the screen_**, on which a long time shine the **_same pixels_**. This is the real cause of "burnout" / "fading"/ etc.

In the normal usage scenarios your screen will be turned off almost all time and lights for a few seconds/minutes when you are want some steam ^^.

I have several devices which are already 2+ years old, and all have the screen in perfect condition.
I just never charge the battery through the devices ;)

ETPNV13, 2017-05-14:
I never charge over USB either and from everything I can find about OLED screens they can last anywhere from 15000 to 100000 hours of "screen on" time so considering the screen is only on maybe a total of 3-8 hours a day that equates to 1000 to 12000 days or 3 to 33 years of use...I guess my only remaining question is do you advise running the screens at 100% contrast or a more moderate contrast setting?

ReMicroN, 2017-05-14:
I have used RS2/3 for about 4 month with 70% of brighness. Now I flash AF firmware and I can see old firmware screen shodow) There's nothing to be done about it.(

GnFlala, 2017-05-14:
auto stealth while charging please. I burned my screen because of this. And no i am not charging but i am monitoring and i am forgeting the device connected on the pc.

ZenFree, 2017-05-14:
Screen timeout settings while charging would reduce if not eliminate this burnout issue.

ETPNV13, 2017-05-14:
@ReMicroN  How often do you use your mod and do you use it a lot outside? I only ask because I have multiple joyetech mods that are over a year old and none of them have screen "burn in" yet....do you charge over USB? do you have screen contrast all the way up? I do not doubt that it has happened to you as I have read about it happening to others as well I am just trying to figure out how it happens, as I said I have multiple joyetech/eleaf/wismec mods but I have 20 of them and I rotate them out as they are different colors/styles/shapes/battery configs so I can switch it up from time to time and I want to avoid whatever people are doing to make this happen as I want to keep these mods for some time as a collection of functional mods lol

ReMicroN, 2017-05-14:
I don't charge it via USB - only external charger. I use myEvic firmware without stealth mode. Brightness was 65-75%. I think the main problem was in excessive screen brightness. I recommend not to set it more than 45%.

ETPNV13, 2017-05-14:
I like being able to read the screen in direct sunlight but if it is going to make my screen start dimming or cause image burn-in I think 15% contrast is good for me...thanks for the response, do you know what "original" contrast setting is in OEM firmware? I read somewhere that it is 17% but now I can not find that post again lol

GnFlala, 2017-05-15:
have the same problem with logo burned in permanently. Sun here is ultra stong and i was using 55% to 65%. 
Now i am trying to reverse the effects with invert the same logo.

TBXin, 2017-05-16:
> Now i am trying to reverse the effects with invert the same logo.

In this case you will "burn" inverted pixels. "darken" pixels cannot be repaired. Only display replacement will help.

GnFlala, 2017-05-16:
oh damn..... Thanks for the help tho man.