maelstrom2001/ArcticFox #376: Dual Pico 200w Firmware Update Issue

Mu5tang71, 2017-05-15:
I just updated my firmware from eleaf web page to (Pico_Dual_V1.04.bin) and now it appears that my charge status is not working... Not sure if it's even charging.. Is there any way to roll back to the factory firmware version not sure what that was either... I am pretty pissed that this happend... It been charging for hours and the staus wont move past 1/3.... Tried taking out batteries and turning back on but no change... Don't they test these updates before releasing them to the public. Contacted customer support and no response... Thanks

ETPNV13, 2017-05-15:
How does this relate to Arctic Fox???

TBXin, 2017-05-15:
@ETPNV13 never mind, seems to be we are some kind of Joyetech/Eleaf/Wismec support crew 😂😂😂

@Mu5tang71 seriously, what kind of support did you expect to get HERE?
We have nothing to do with official firmware. We are developing 100% custom firmware.

ETPNV13, 2017-05-15:
On a side note...charging via an external charger would completely resolve this issue...just a thought

Mu5tang71, 2017-05-15:
@ETPNV13 

Well, I have seen in a previous [post](https://github.com/maelstrom2001/ArcticFox/issues/46) that someone got some help fixing an issue similar to mine using the same exact mod... so I figured it couldn't hurt to post a new Issue maybe someone could actually help me without coming at me all sideways like I am posting in a foreign language... As in the previous post, someone gave them a confirmed working firmware and was able to help them. But that firmware is no longer there or available for download... 

On that side note: How does an external charger fix the issue with the status showing false power display readings. The whole point of the post was to get that working properly.  I think it's charging the unit but not displaying proper current power levels.

Anyway...... nevermind I don't want to stress you out any further by posting in the group as I can plainly see that you are a bit confused not even knowing what brands the forum supports and the reson why I posted in here.

TBXin, 2017-05-15:
@Mu5tang71 message that you indicated about Pico Dual support by **our firmware** not original firmware by Eleaf.

All this "forum" is about out firmware and its support.

Mu5tang71, 2017-05-15:
I don't care what firmware it is or was as long as it worked properly... lol

closed 2017-05-15 by TBXin

Mu5tang71, 2017-05-15:
This guy asked the same exact [question](http://i134.photobucket.com/albums/q92/mu5tang/Misc/1_zpswylqp1oq.jpg) why don't you go and bash him for asking for a stock firmware

TBXin, 2017-05-15:
@Mu5tang71 he was asking about strange behavior in the **nightly build of the ArcticFox** firmware, not in the original firmware...

maelstrom2001, 2017-05-15:
@Mu5tang71 Are you looking the stock firmware? It is Pico_Dual_V1.04.bin, it is the only, it is available for download from Eleaf and AFAIK it was not changed since December 2016.

ETPNV13, 2017-05-15:
@Mu5tang71 Okay, to be clear...no one "came at you sideways", we all typed a valid response. You asked about stock firmware in an issues section for completely unrelated firmware...this is not any different than asking on a linux issues forum about a windows problem, just because the hardware is the same does not mean your issue is relevant here, so my question was quite reasonable. In regards to me suggesting an external charger, you never stated that the batteries were charging and only the indication of the charge was wrong...from what you wrote it appears as if you were having issues with it not charging at all so I gave what seemed to be a reasonable suggestion considering the small amount of information you provided...if you didn't want suggestions then why did you bother posting here at all???

ETPNV13, 2017-05-15:
@Mu5tang71 There are a huge multitude of other forums specifically for the questions you have or even possibly a quick google search might be of some help as well...www.letmegooglethatforyou.com