maelstrom2001/ArcticFox #377: Joyetech egrip II

Andy2vape, 2017-05-15:
Hi, I tried to update my device --- device ID 
[E083] Joyetech eGrip II
ArcticFox from 170201 is running perfectly, 
By starting updating, the egrip II is recognized, but when I´m start flashing from file, 
an error message coming "Opened Firmware File is not suitable fro the connected device."
Any ideas ? 
Thx.
Andy

TBXin, 2017-05-15:
Hello,

Use latest NToolbox.

Andy2vape, 2017-05-15:
Hello, I use version 2.0
NFirmwareEditor-v7.0.7
Problem seen with this version, yesterday I updated other device, no problem, runs perfectly ?
Thx for investigation
Andy

TBXin, 2017-05-15:
@Andy2vape 

What devices did update yesterday and what version of the ArcticFox did you install? 
What version of the NToolbox did you use to install ArcticFox to these devices?

The more information you provide, the easier it will be to tell you what the matter is.

Andy2vape, 2017-05-15:
I´ve updated yesterday a RX200S to the newest firmware 170510, works fine, very good software.
I´m using Version 2.0, NToolbox v2.0 and ArcticFox 170510, I´ve tried the older relase also 170509, same error message.
The egripII is running currently with ArcticFox 170201

The only difference I´m using another PC... maybe this causing the problem.
My PC is Windows 10, newest updates on.

Andy2vape, 2017-05-15:
[image]
[image]
[image]

Andy2vape, 2017-05-15:
I reverted to Original Firmware to get a clean start, but same messages. Hope this is helping to reproduce.
Thx Andy

TBXin, 2017-05-15:
> The only difference I´m using another PC... maybe this causing the problem.

Yeah. Try to check this Windows setting https://github.com/maelstrom2001/ArcticFox/issues/330#issuecomment-300304910

Also you can update .NET Framework.

Andy2vape, 2017-05-15:
Hello Again, I tried another PC and it´s working.
 I don´t understand why this is happening ? I will try on another PC to reproduce the setting of PC

Andy2vape, 2017-05-15:
YES, maybe it´s .NET Framework

Andy2vape, 2017-05-15:
I think we can close this, was my PC somehow wrongly configured. All other PC´s are working where I tested.
Thanks for your support !

TBXin, 2017-05-15:
No problem! Have a nice day.

closed 2017-05-15 by TBXin