maelstrom2001/ArcticFox #379: Serious battery sag... new firmware or new batteries?

traviswparker, 2017-05-15:
RX2/3 in 2-cell mode. I recently installed 20170510 and also started using a brand new pair of Smasung 25R cells. When my charge level gets down to 4.0v (80%) I start to get the "Low Battery" message and the mod will not fire. Ni 200, 0.11, 450F, max 150W. I'll start to get "weak battery" around 85%.

I'm confused my new 25Rs can't provide 150W at 4.0v. Doing the math, that should just be < 20A per cell, well within the specs for a 25R. The batteries they replaced were also 25Rs, and despite being worn out and holding far less charge than they used to have, I wouldn't even get "Weak Battery" until they were down below 25%.

Attached is a screenshot from device monitor when I request a puff and "Low Battery" appears. Is that extreme voltage drop normal? Did something in the firmware change or did I buy defective batteries?
[image]

TBXin, 2017-05-15:
Hello,

Let's do some math.
To get 150 watts for 0.11 Ohm coil(s) DC/DC converter should provide 36.92A and 4.06V.
And what is fed to the input of the converter? 2 serial connected batteries. 
I can suggest that efficiency of the DC/DC converter is about 85%.

So, we need to feed about 172 (150+15%) watts to the converter to get 150 watts for coil(s).
As i can see your batteries under load have about 2.9 volts. 
Using Ohm Law we can calculate how much current we need from batteries to get desired 172 watts.
I = P / V
172 / (2.9 + 2.9) = 29.65A from each battery.

I believe that your batteries cannot provide so much current with their current charge level.

maelstrom2001, 2017-05-15:
I think it is bad batteries, huge voltage drop. Got on Primo with 25R:

[image]

From 3,60 to 3,00 volts under load, not bad I think.

traviswparker, 2017-05-15:
I think you might be right. I don't know if they are actually bad or if these only hit the 20A rating they are supposed to. My old ones must have been better than advertised. From what I can gather, 20A is the minimum a 25R is capable of but some can actually go much higher.

closed 2017-05-15 by traviswparker