maelstrom2001/ArcticFox #380: Power Limit

snclab, 2017-05-15:
Hello,

I can remove power limit ?
can add this feature ?

Thanks

TBXin, 2017-05-15:
Hello,

No.
For what?

Dil3mm4, 2017-05-15:
## Limitations

The safe use of our software it's one of our main goals. We retain all factory restrictions, because the device manufacturer did NOT make them "just for fun". What are the limitations?

-  Minimal Resistance (For all supported devices)
      * 0.05 Ohm for Temperature Control mode
      * 0.1 Ohm for Vari-Watt mode

-     Output Wattage (Depending on the device)

-     Maximum Charging Current (Depending on the device)

-     Maximum Output Current (Depending on the device)

-     Output Voltage Range (Depending on the device)

"Depending on the device" means that each device has its own values, which are choosen by the manufacturer. You can find this values on the product page of your manufacturer's website.

**All requests to change these values will be rejected as we don't want you to hurt yourself or your device.**

Readme helps sometimes.

Dil3mm4, 2017-05-15:
@TBXin I think that could lead to counterfeit devices power specs to build "identical" clones on the sheet...at the expense of users safety.

snclab, 2017-05-16:
With the original Cuboid firmware thanks to the patches in VFirmwareEditor I can unlock the power limit
This is convenient when the box locks watts with some resistances

TBXin, 2017-05-16:
Maximum output power is depending on the resistance, output voltage and output amperes.
Cuboid 150/200W has limitations: 9V / 25A / 200W.
You can use any Ohm law calculator to calculate the maximum power depending on your resistance.

snclab, 2017-05-16:
thanks, but then why is there a patch to remove this in NFirmwareUpdater?
This only works with the original firmware

TBXin, 2017-05-16:
Remove what?

snclab, 2017-05-16:
power limit
there is an patch to remove power limit with original firmware with NFirmwareUpdater

TBXin, 2017-05-16:
There are no such patch.

TBXin, 2017-05-16:
Anyway. "Limitations" section in the readme already describes our position.

closed 2017-05-16 by TBXin

snclab, 2017-05-16:
Ok thanks, but if you see patches for Cuboid on NFirmwareUpdater, it is present