maelstrom2001/ArcticFox #382: screens not centered

cdyseplc, 2017-05-16:
all the screens for all 3 themes are not centered, is there a way that i can center them via the toolbox or something?

m600x, 2017-05-16:
they are all centered, no.

TBXin, 2017-05-16:
What do you mean?
You can take a screenshot and count the pixels.

cdyseplc, 2017-05-20:
will do that and get back soon

cdyseplc, 2017-05-20:
[image]
[image]
[image]
[image]
[image]

here are the screens!

joeblowma, 2017-05-20:
You can blame wismec for that, where they mounted the screen there is no guide to hold the screen centered/straight (a few people have gotten screens at an angle), it's just stuck there with some double sided tape and the ribbon soldered down to the PCB.

Mine is set a little too high up but relatively centered, and really far back from the glass - it might have worked out perfect if they had used thicker tape.

cdyseplc, 2017-05-20:
its not actually if you see the first screen, it is off by maybe one or two pixels

cdyseplc, 2017-05-20:
[image]

TBXin, 2017-05-20:
Actually you can't draw pixel perfect circle with a width/height of a multiple of 2. Thats why you can see 1 pixel gap from the left.

cdyseplc, 2017-05-20:
okok its ok then =)

closed 2017-05-20 by TBXin