maelstrom2001/ArcticFox #383: Screen Stealth mode in Power Bank Mode  support request

ilyxa, 2017-05-16:
While running box (Cuboid 200, Pico 75W) in Power Bank mode, screen is on all time which is not good for OLED display. Can you please add support for stealth mode in Power Bank mode?

ZenFree, 2017-05-16:
Personally, I would prefer settings for screen timeout instead.

ilyxa, 2017-05-16:
Not important, just turn screen off in any manner ;)

maelstrom2001, 2017-05-16:
There is no problem to turn off display completely, but I personally prefer to see in which state box is right now. If it works (Power Bank output is on), it should indicate it in any way. Now used 0% brightness level instead of user defined, I think it should not lead to OLED degradation dramatically. Eventually, it is a matter of taste and a point of view. We do not reject categorically, but we'll not promise anything either. Thanks for understanding.

maelstrom2001, 2017-05-17:
https://nfeteam.org/forum/threads/arcticfox-170517.30/

closed 2017-05-18 by maelstrom2001