maelstrom2001/ArcticFox #384: Foxy skin: Live battery voltage on bottom left battery meter?

Justin-L-Franks, 2017-05-16:
Is there any reason why the battery voltage display at the bottom left doesn't show the voltage drop when firing? Both the Classic and Circles skin do, and this is pretty useful.

messywrk, 2017-05-17:
+1, I also liked the live voltage while firing.

maelstrom2001, 2017-05-17:
You can use battery voltage line to display it while firing:

[image]   [image]

Justin-L-Franks, 2017-05-17:
There are already three separate battery meters (#356), I don't want a fourth. I'm asking if there is a technical reason why the battery voltage displayed at the bottom of the Foxy skin behaves differently than on the other skins.

maelstrom2001, 2017-05-17:
> three separate battery meters

In standby. While firing there will be displayed the lowest battery voltage under load, in runtime, screenshots are above.

> if there is a technical reason why

Why we should make all skins similar to each other? :)

Justin-L-Franks, 2017-05-17:
> In standby. While firing there will be displayed the lowest battery voltage under load, in runtime, screenshots are above.

Yes, but you have to give up another option to show it. And then you would have four separate battery meters, which is pretty pointless.

> Why we should make all skins similar to each other? :)

Because it's a rather useful feature, which also frees up space for other useful data.

maelstrom2001, 2017-05-18:
All this is a matter of taste. It is impossible to please everyone. Please do not forget that first of all the firmware is made for personal use. So you can discuss as much as you want, but that does not mean that absolutely all wishes will be fulfilled. I hope for understanding and thank you for using our products.

Justin-L-Franks, 2017-05-18:
I understand completely. It just seems odd that you would remove a useful feature from the skin that people like. And if you didn't think it was useful yourself, you never would have put it in the other skins in the first place.