maelstrom2001/ArcticFox #385: Battery indication in Power Bank mode

ilyxa, 2017-05-16:
Checked on Pico 75.
Battery indication "stuck" in fully charged state all time while power bank was on (load current was about 0.3A if this is important, Avatar RC connector is in use). 

Problem: The battery indicator did not change until the box was completely turned off. Looks like this is bug in UI?

maelstrom2001, 2017-05-16:
Thanks for report, yep, possible bug. Will take a look.

ilyxa, 2017-05-16:
Also checked on Cuboid 200 - same problem.

maelstrom2001, 2017-05-16:
Please test attached build. Note that under load (while Power Bank is on) battery indicator will display battery level lower than in standby. It is normal, 'cause battery voltages will drop a little under load.

[af_170517.zip](https://github.com/maelstrom2001/ArcticFox/files/1006067/af_170517.zip)

ilyxa, 2017-05-17:
Work as expected, checked on Pico 75.
[image]
[image]

Thank you!

closed 2017-05-18 by maelstrom2001