maelstrom2001/ArcticFox #386: Our forum is open

TBXin, 2017-05-16:
Hello here!
We are happy to announce that our forum is ready for our dear community 😉
We will gradually fill it with content 😃

Please do not create new issues on GitHub. We will close this possibility in a 2-4 weeks. 
If you have questions or suggestions or you just want to talk - your are welcome! 

https://nfeteam.org/forum. 

We are hoping that you will like it ;)

dannysmobiles, 2017-05-17:
I've tryed signing up and I'm not getting this message but I've had no email did it yesterday Your account is currently awaiting confirmation. Confirmation was sent to

maelstrom2001, 2017-05-17:
@dannysmobiles your account has been activated, welcome!

ETPNV13, 2017-05-19:
@maelstrom2001  can you approve mine as well, my user name is invisiblehand13 and I have resent the email confirmation like 3 times and still no approval :(

TBXin, 2017-05-19:
@ETPNV13 done!

TBXin, 2017-05-19:
For some reason @live and @outlook are blocking activation emails from our message server.

ETPNV13, 2017-05-19:
@TBXin Perfect! thanks!!