maelstrom2001/ArcticFox #387: Battery voltage in displayed inverted colors on charge screen

malinges, 2017-05-16:
Hi,

I've been charging my Predator with USB, and noticed that sometimes, one of the batteries' voltage is displayed in inverted colors:
[image]
What does that mean?
Is this somehow related to balanced charging?

BTW, great work guys! ArcticFox is getting better and better ;-)

TBXin, 2017-05-16:
Hi,

This is how "balancing" works. Inverted battery means that this battery is waiting until other battery voltage is not become same.

For future read this please: https://github.com/maelstrom2001/ArcticFox/issues/386

malinges, 2017-05-17:
Ok, that's what I thought, but I wasn't sure.
I'll also be keeping an eye on the forum...
Thanks!

closed 2017-05-17 by malinges