maelstrom2001/ArcticFox #388: My brand new Primo 2.0 reads "aromizer short" whereas my old Primo works fine

javak810i, 2017-05-18:
Hey, so I have two Evics Primo, the old version and the new 2.0 which just arrived yesterday. Both have the latest 510 build of Arctic Fox Installed. I'm, using the Pharaoh RTA with staggered clapton coil reading at 0.29 Ohm. On the older Primo everything works fine. But on the Primo 2.0 I get "atomizer short" message and 0.00 Ohm readings after pressing the fire button. Other tank (tobeco super tank mini with 0.5ohm coil head) works fine on the primo 2.0.

What could be the issue?

maelstrom2001, 2017-05-18:
Hmmm... let me guess: is Primo 2.0 black painted? :smile:

javak810i, 2017-05-18:
@maelstrom2001 Nah, it's the silver version :)

javak810i, 2017-05-18:
@maelstrom2001 Well, my bet is on the second option as well, because the resistance on the coil I'm using in my Pharaoh RTA is 0.29 Ohm and it works great on the older Primo "1.0" using the same build of Arctic Fox.
I guess I'm just gonna flash the original firmware and sell it :D.
P.S. I noticed that the finish on the silver Primo 2.0 is a bit different to the silver Primo 1. On Primo 2.0 it's similar to the Wismec Predator finish - more rough and easier to scratch/polish.

maelstrom2001, 2017-05-18:
> it works great on the older Primo "1.0" using the same build of Arctic Fox

That is the answer. Original firmwares for early Primo and 2.0 are almost identical (with minor differences in charging current and max. power output). So the problem in your new 2.0 is hardware, something wrong with 510 connector. We never had so much reports on "Atomizer short" messages until the new generation of devices was released. Quality of new boxes is really poor. Maybe there is some tricks in original firmware that helps to hide the issues, but... on really good hardware AF works absolutely normal.

P. S. Same here, I never saw "Atomizer short" on my early Primo.

javak810i, 2017-05-18:
@maelstrom2001 Yeah, sad times. 
I remember my first mod - eVic VTwo. It was supposed to be quite shitty but I never had any issues with it, even when using with myevic and arctic fox FW. 

Thinking... what are some other affordable (under 50USD) dual 18650 box mods with balanced charging that have no 510 pin issues? :D We know Preador is a mess, Primo 2.0 hasn't started great... I kinda like the Fuchai 213 Plus but it's not very cheap and doesn't support your great firmware ofc :)

maelstrom2001, 2017-05-18:
[offtopic] :smile: 
Fuchai 213 / Plus can be bought because of another custom firmware, [Vortex Mod](http://xvtx.ru/ecig/). Didn't see and test it, but I can say that the author made great work :+1:

javak810i, 2017-05-18:
@maelstrom2001 
[offtopic]
Yeah but it needs to he HW modded if you want to install it, so bye bye warranty. Anyway I heard it's mainly useful if you want good TC and I don't use TC at all :).