maelstrom2001/ArcticFox #392: Screen doesnt time out when USB charging (RX2/3 Latest AF Build)

kyan31, 2017-05-19:
I have Arctic Fox installed on my RX2/3, Build 170510 and when I charge the device via USB the screen never times out so it is always on while charging. This may seem like a small issue but the RX series use OLED screens which are susceptible to screen burn-in. I have checked all the time out settings and I cant find any option to time out the screen while charging. Normally I use a decicated 4 bay battery charger to charge, but sometimes I find myself in situations where USB is the only option so hopefully the timeout while charging thing can be implemented in a newer build. There is a workaround however. If you plug in the mod, then while plugged in switch to stealth mode and then fire the mod for like a millisecond, the screen will go off but still be charging. This isn't very streamlined though but it works. Anyway if the timeout feature was implemented that would be very handy because the firmware is perfect besides this minor issue. Thank you!

TBXin, 2017-05-19:
Hello, 

This is default behavior even in the OFW. For some reason many people likes to watch how they screens are dying :) more smarter users using stealth mode or charging batteries in the external charger.

kyan31, 2017-05-19:
@TBXin Haha that's true 😂. Thing is though on the original wismec firmware the screen timed out after like 5 seconds, so the implementation is possible. But yeah I think I'll just deal with it as it's a small issue anyway, thank you for the reply.

TBXin, 2017-05-19:
@kyan31 hmm. I've never seen that any device from wismec turn off display when they are charging. When charging is done - yes, but not when charging in process.

kyan31, 2017-05-19:
@TBXin the RX200S/2/3 both did, they would show the basic charging indicator for like 5 seconds and then the screen would switch off.

TBXin, 2017-05-19:
Only when batteries are charged, otherwise charging screen will appear until charge completed.