maelstrom2001/ArcticFox #395: Flip screen when firing only

zviratko, 2017-05-19:
This is a feature request

Can you implement an option to flip the screen only when firing?

Thanks

TBXin, 2017-05-19:
Hello,

You can adjust puff screen delay to keep last puff info on the screen for a specified time.

zviratko, 2017-05-20:
That's not what I am talking about at all :-)

There is an option to flip the screen upside down. Great if you want to see what's going on on the display when vaping, but awkward when not. It would be great to have an option to have the screen flipped upside down when vaping, but normal when not vaping.

TBXin, 2017-05-20:
I am understood your case and that's why I suggested to you use delayed puff info disappearance.

All relative info lines, power, temperature, this all will be on a screen for a certain time after you finish puff.

zviratko, 2017-05-20:
Viewing live data is completely different to seeing the last snapshot...